ananevilya/Arduino-ST7789-Library #1: Refresh

C4PT41ND34DP00L, 2018-08-14:
Is there a way to refresh the screen without having to fill it with black?

ananevilya, 2018-08-14:
What do you mean? You can fill screen with any color that you want. Filling with custom color need for "erasing" all object. You can use a little effortless way - filling custom region with objects on screen that you need to remove.

C4PT41ND34DP00L, 2018-08-15:
Yeah but that causes it to flicker was looking for something like the clearDisplay function of other libraries as this seems to be the only library that works for this oled.

ananevilya, 2018-08-16:
This is a very high dpi display. It have 240x240 pixels (each pixel have 2 bytes of color), that's mean we have more than 115kB to send for clearing screen. It's not possible to send big amount of data very fast. You can try to increase SPI speed, but it may be not workable (if you have long wires, flux residues, etc.).
What platform do you use?

C4PT41ND34DP00L, 2018-08-16:
I am using an Arduino Pro Mini on a breadboard with short wires.

ananevilya, 2018-08-16:
Try to change **SPI_CLOCK_DIV4** to **SPI_CLOCK_DIV2** at line 203 in **Arduino_ST7789.cpp** file.
It should be two times faster.

remixvit, 2018-11-21:
Hello. I can’t use this library with STM microcontrollers in arduino.
"Arduino_ST7789.cpp:56:37: error: expected primary-expression before 'void'

 #define SPI_BEGIN_TRANSACTION()    (void)"

ArsenioDev, 2018-12-19:
Remixvit, you need a transactional SPI, the rogerclark core is not supporting that yet.