ananevilya/Arduino-ST7789-Library #2: Does not work with the 1.3 ST7789 240*240 displays

ArsenioDev, 2018-09-07:
Tested this library with my 1.3 displays, never switches off of the default black. These do not have the CS pin broken out 
https://www.ebay.com/itm/1-3-Inch-IPS-Full-View-HD-LCD-Display-Module-ST7789-SPI-240-240-12Pin-3-3V-4xSPI/173396379538

ananevilya, 2018-09-07:
1.3 display without CS pin (like from ebay or aliexpress) work very well with this library. Please check wiring.
What type of initialization do you use?

ArsenioDev, 2018-09-07:
I've tried the hardware SPI and software both with and without the CS pin, no dice on either attempt. Got any ideas?

ananevilya, 2018-09-10:
May be broken screen / arduino. You can try to check signals on all display pins with oscilloscope or logic analyzer.

ArsenioDev, 2018-09-10:
Known good arduino, will check on different platforms and take a check on the signals

C4PT41ND34DP00L, 2018-09-11:
I have this exact display and it works well.

dreamy1, 2018-09-21:
Hello,

i have exact the same problem with not responding lcd module - i don't see a reaction and the module stays black. Tested it with another display and it is the same.

Alle wires are correct connected and the settings are correct (i have a lot of experience with adapting TFT modules).

Is it possible that the µC is the problem? I have an Pro Mini 3.3V/8MHz attached, maybe the frequency of 8Mhz is a problem for the SPI transactions?

Any help appreciated, thank you very much in advance!
Stefan

dreamy1, 2018-09-22:
Problem solved, it was a bad contact on the reset line.

The lib works perfect, many thanks for your work!

ArsenioDev, 2018-09-26:
Ran flywires from the flat flex to pins, still no dice. Tested on a teensy, 8266, leonardo, stm32. no dice on any

ananevilya, 2018-09-27:
I think you have broken screen. Try with another one. This is not software problem.

closed 2018-09-27 by ananevilya

ArsenioDev, 2019-01-13:
Tested with different screen, noted silkscreen DC/RST are swapped and now have all working on an 8266, same code with hardware SPI pins does however not work on a Teensy

taka-tuos, 2019-04-23:
I am also trying to use a 240x240 IPS LCD module (obtained from AliExpress, without CS pin) with the ST7786 in the Maple Mini.
It worked with software SPI but not with hardware SPI.

2ni, 2020-01-17:
> I am also trying to use a 240x240 IPS LCD module (obtained from AliExpress, without CS pin) with the ST7786 in the Maple Mini.
> It worked with software SPI but not with hardware SPI.

It worked for me when I changed SPI_MODE2 -> SPI_MODE3 in the code of Arduino_ST7789.cpp (got the hint from [here](https://github.com/Bodmer/TFT_eSPI/issues/163#issuecomment-467220399)

omerk, 2020-08-28:
For future Google searches: On the modules I have here without the CS pin exposed (presumably grounded), using `tft.init(240, 240, SPI_MODE3)` made them work with this library. Reference: https://github.com/Bodmer/TFT_eSPI/issues/163

witnessmenow, 2020-12-04:
I wasn't able to get this working with an ESP8266. I get some garbage on the screen followed by a black screen, but it doesn't display anything after that.

> 
> 
> For future Google searches: On the modules I have here without the CS pin exposed (presumably grounded), using `tft.init(240, 240, SPI_MODE3)` made them work with this library. Reference: [Bodmer/TFT_eSPI#163](https://github.com/Bodmer/TFT_eSPI/issues/163)

Tested today, this does not work. 

`no matching function for call to 'Arduino_ST7789::init(int, int, const uint8_t&)'`

This library inherits from AdafruitGFX, the adafruit version of this library inherits from Adafruit_SPITFT, which does support the above.

For other future reference, I was not able to get this working with the Adafruit one either, LadyAda suggested that it will not work since there is no CS. 

Works great with tft_espi though

anabolyc, 2021-12-22:
Had the same issue, `tft.init(240, 240, SPI_MODE3)` did the trick