ananevilya/Arduino-ST7789-Library #3: Does not work with 1.54" IPS SPI TFT (ST7789V)

dreamy1, 2018-10-13:
Hello,

after testing the library with the 1.3" IPS Screen with ST7789 (works very good!), i have problems to get the library working with a 1.54" IPS Screen with ST7789V controller.:
http://www.ecyberspaces.com/productsview.asp?id=9743

The only difference is a CS pin onto the module - the library has this option already and i have activated it, but the screen stays black.

This is the code i use:
***********************************************************************************
#define TFT_DC    7
#define TFT_RST   5 
#define TFT_CS    10 // only for displays with CS pin
#define TFT_MOSI  11   // for hardware SPI data pin (all of available pins)
#define TFT_SCLK  13   // for hardware SPI sclk pin (all of available pins)

//You can use different type of hardware initialization
//using hardware SPI (11, 13 on UNO; 51, 52 on MEGA; ICSP-4, ICSP-3 on DUE and etc)
//Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST); //for display without CS pin
Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST, TFT_CS); //for display with CS pin
//or you can use software SPI on all available pins (slow)
//Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST, TFT_MOSI, TFT_SCLK); //for display without CS pin
//Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST, TFT_MOSI, TFT_SCLK, TFT_CS); //for display with CS pin
************************************************************************************************

Any ideas? Help would be appreciated to get this lcd module work, thanks in advance!

Regards,
Stefan

dreamy1, 2018-10-22:
The ST7789 und ST7789V is the same controller. What i found out: the module in my link is a 3-wire SPI 9bit module without D/C line...and this should be the problem. Your library is support 8bit lcd only, right?

ananevilya, 2018-10-23:
Yes, my library support only 8bit mode. Because arduino SPI (atmega328p hardware) support only 8bit SPI data length. You need to develop your custom software 9bit SPI. And than change "writecommand" and "writedata" functions in my library.
I can help you if you want.

dreamy1, 2018-10-23:
That would be great!

I found a 9bit library for ST7735...it should be easy to adopt it: https://github.com/shuzonudas/ST7735-9bit

I will have a look if there are other libraries for 9bit which support adafruits gfx library - this would be the best solution for me because i use it already with a tft module which will be replaced by the IPS screen in the link above.

ananevilya, 2018-11-05:
I added branch spi9bit for you to testing 9bit mode. https://github.com/ananevilya/Arduino-ST7789-Library/tree/spi9bit

dreamy1, 2018-11-05:
Wow, thank you very much!!!

I will test it and give you feedback - big thanks for your help again!

dreamy1, 2018-11-09:
Hello again,

today i had some time to test the new 9bit branch:

The good news is: the library is in pricible working well, so the example garphicstest sketch is visible on the screen. Thanks a lot, this is a really big progress to get this module working!

But there are some minor issues:
- the screen seems to be initialized only for the upper 4/5 of the screen, so the sketch is only there visible. The lower side of the display is disturbed. It seems that the "zero point" for starting at 0,0 is about 50 rows to high because the pictures in the sketch are postponed for this area.
- the sketch is very slow compared with the "normal" one in 8 bit. Is there a chance to get this accelerated?

Thanks again for your great help! I am sure that we are very close to get this module 100& working :-)

Warm regards,
Stefan

dreamy1, 2018-11-09:
Update:

with follwowing modification in the ST7789.h the first issue is solved:

#define ST7789_240x240_XSTART 0
#define ST7789_240x240_YSTART 40

dreamy1, 2018-11-10:
I tested it with Arduino Pro Mini on 8Mhz. Maybe there is an increase of speed on 16Mhz (Arduino Uno etc.) for software spi.

Is it possible to get 9bit working with hardware spi?

ananevilya, 2018-11-10:
Yes, faster board (with 16Mhz) will double the SW SPI speed.
You can also try Arduino DUE.
AVR HW SPI can work only in 8bit mode.

dreamy1, 2018-11-10:
No problem - it is absolutely perfect now for my purpose and the library works very good!

Thanks again for your help!

closed 2018-11-10 by dreamy1

deirvlon, 2019-05-20:
Hello

I want to buy this display 
[https://www.aliexpress.com/item/32920192414.html?gps-id=pcStoreJustForYou&scm=1007.23125.122752.0&scm_id=1007.23125.122752.0&scm-url=1007.23125.122752.0&pvid=c7303b18-38d6-4b3c-abf9-50dca97946bf&spm=a2g1y.12024536.smartJustForYou_68431350.3](url)

How can i connect and use this display with any Arduino? 
[image]

I read all of your conversation and match the pins of displays and i saw that this display above have not DC pin . It is problem?

**/***************************************************
  This is a library for the ST7789 IPS SPI display.
  Written by Ananev Ilya.
 ****************************************************/

#include <Adafruit_GFX.h>    // Core graphics library by Adafruit
#include <Arduino_ST7789.h> // Hardware-specific library for ST7789 (with or without CS pin)
#include <SPI.h>

#define TFT_DC    8
#define TFT_RST   9 
#define TFT_CS    10 // only for displays with CS pin
#define TFT_MOSI  11   // for hardware SPI data pin (all of available pins)
#define TFT_SCLK  13   // for hardware SPI sclk pin (all of available pins)

//You can use different type of hardware initialization
//using hardware SPI (11, 13 on UNO; 51, 52 on MEGA; ICSP-4, ICSP-3 on DUE and etc)
//Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST); //for display without CS pin
//Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST, TFT_CS); //for display with CS pin
//or you can use software SPI on all available pins (slow)
//Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST, TFT_MOSI, TFT_SCLK); //for display without CS pin
//Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST, TFT_MOSI, TFT_SCLK, TFT_CS); //for display with CS pin
Arduino_ST7789 tft = Arduino_ST7789(-1, TFT_RST, TFT_MOSI, TFT_SCLK, TFT_CS); //for display with CS pin and DC via 9bit SPI**

_Arduino_ST7789 tft = Arduino_ST7789(-1, TFT_RST, TFT_MOSI, TFT_SCLK, TFT_CS); //for display with CS pin and DC via 9bit SPI_ In this line you did not use DC pin WHY?

Please help me!

Regards,
Kamran

ananevilya, 2019-05-21:
> Hello
>
> I want to buy this display [...]
Hello. No, it's not a problem. I think this display uses 9-bit SPI mode, where DC is 9th bit of data.

deirvlon, 2019-05-21:
Aaa i understand Thank You so much . 

https://github.com/ananevilya/Arduino-ST7789-Library/tree/spi9bit

May be this will be the right library , am i right?

ananevilya, 2019-05-21:
> Aaa i understand Thank You so much .
> 
> https://github.com/ananevilya/Arduino-ST7789-Library/tree/spi9bit
> 
> May be this will be the right library , am i right?

Yes, you are right. This branch for 9bit SPI mode.

deirvlon, 2019-05-21:
Thank you so much again)

studiokoenig, 2020-01-18:
Hey there,
im using your library for a university project and used a Arduino Mega to prototype.
It worked perfectly.

Next step was to run the same code on a teensy 4.0 and I got some errors I couldn't resolve:

Arduino: 1.8.10 (Mac OS X), TD: 1.49, Board: "Teensy 4.0, Serial, 600 MHz, Faster, US English"  graphicstest: 

In function 'void setup()': graphicstest:47: warning: ISO C++ forbids converting a string constant to 'char*'     testdrawtext("Lorem ipsum dolor sit amet, consectetur adipiscing elit. Curabitur adipiscing ante sed nibh tincidunt feugiat. Maecenas enim massa, fringilla sed malesuada et, malesuada sit amet turpis. Sed porttitor neque ut ante pretium vitae malesuada nunc bibendum. Nullam aliquet ultrices massa eu hendrerit. Ut sed nisi lorem. In vestibulum purus a tortor imperdiet posuere. ", WHITE);                 
                                                                                                                                                                                                                                                                                                                                                                                      **^ /Users/XXX/Documents/Arduino/libraries/Arduino-ST7789-Library-spi9bit/Arduino_ST7789.cpp: In member function 'void Arduino_ST7789::commonInit(const uint8_t*)': 

/Users/XXX/Documents/Arduino/libraries/Arduino-ST7789-Library-spi9bit/Arduino_ST7789.cpp:228:13: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'RwReg* {aka volatile unsigned char*}' in assignment    dcport    = portOutputRegister(digitalPinToPort(_dc));              ^** 

/Users/XXX/Documents/Arduino/libraries/Arduino-ST7789-Library-spi9bit/Arduino_ST7789.cpp:231:12: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'RwReg* {aka volatile unsigned char*}' in assignment   csport    = portOutputRegister(digitalPinToPort(_cs));             ^ 

/Users/XXX/Documents/Arduino/libraries/Arduino-ST7789-Library-spi9bit/Arduino_ST7789.cpp:260:17: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'RwReg* {aka volatile unsigned char*}' in assignment      clkport     = portOutputRegister(digitalPinToPort(_sclk));                  ^ 

/Users/XXX/Documents/Arduino/libraries/Arduino-ST7789-Library-spi9bit/Arduino_ST7789.cpp:261:17: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'RwReg* {aka volatile unsigned char*}' in assignment      dataport    = portOutputRegister(digitalPinToPort(_sid));                  ^ 

Mehrere Bibliotheken wurden für "Adafruit_GFX.h" gefunden  
Benutzt: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Adafruit_GFX  
Nicht benutzt: /Users/XXX/Documents/Arduino/libraries/Adafruit_GFX_Library 

Mehrere Bibliotheken wurden für "Arduino_ST7789.h" gefunden  
Benutzt: /Users/XXX/Documents/Arduino/libraries/Arduino-ST7789-Library-spi9bit 

Mehrere Bibliotheken wurden für "SPI.h" gefunden  Benutzt: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/SPI 

Fehler beim Kompilieren für das Board Teensy

mik00, 2020-04-14:
Hi ananevilya

I have 3 of these displays and for a week I have made no progress driving from an ESP32 - apart from being able to illuminate the backlight. So I have something right at least.  I'd like to swap notes with someone who HAS managed to get it working, and it seems like you must have.  I'm pretty experienced with SPI, and I am using your code (thank you big time), and have now tried every instantiation flavor :-)

I have done continuity tests as far as the tiny pins on the socket, and all seems OK.

I am pretty sure I am generating the correct signals, (SCK, SDA, WR and CS), because the scope output levels(3v3), timings and polarities look convincing. But I'm still not convinced I have wired it up correctly, because I have run out of other ideas.  I have assumed that the legend on the _display's ribbon connector_ are the correct pin designations.  I notice that the Hirose connector specification numbers them in a different order (doesn't it?), but I have gone with the ribbon cable numbers.

What is the simplest wiring config that will work?  Are there any pins that must be tied hi/lo? I've tried a lot of combos.  Would you be willing to share a picture of your connections to the Hirose plug?

Cheers.

dreamy1, 2020-04-14:
Did you apply a 1.8V and a 2.8V power supply voltage for the display? I hope you did not apply 3.3V from your ESP - otherwise this display should be defective.

mik00, 2020-04-15:
Stefan!

Thanks so much for the prompt response.

Well yes, I did connect the 3v3 to the 1v8 and 2v8.  😢 But I read the
Sitronix spec and it said that 3v3 was a safe limit.  Of course I am
playing with fire, so perhaps you have put your finger on my crass mistake.
I'll try some regulators on my one remaining display.
[image: image.png]

Assuming that over-voltage is my problem, can confirm that the pins really
are numbered  *clockwise *on the connector on the ribbon cable, and
correspond to the assignments on the datasheet you posted originally.
[image: image.png]
Confusingly, footprints downloaded from the usual places have them numbered
like this.

[image: image.png]
Thanks for taking the time to answer my questions.
M

dreamy1, 2020-04-15:
I will get the complete datasheet of it today and come back to you. Some displays tolerate 3.3V at the pins above, so there is a chance that it still works.

dreamy1, 2020-04-15:
So, according to the datasheet the 3.3V level on the power supply pins are ok.
You can drive the lcd with a 3-line 9 bit SPI (Pin 7 = WR tied to GND) or with two serial data lines (here WR is used for data). 

So this declaration should be right:
Arduino_ST7789 tft = Arduino_ST7789(-1, TFT_RST, TFT_MOSI, TFT_SCLK, TFT_CS); //for display with CS pin and DC via 9bit SPI

Here is the pin mapping:
TFT_RST --> LCD Pin 10
TFT_MOSI --> LCD Pin 6
TFT_SCLK --> LCD Pin 8
TFT_CS --> LCD Pin 9
and LCD Pin 7 has to be tied to GND.

dreamy1, 2020-04-15:
[image]
This is the connector pin assignment.

mik00, 2020-04-15:
Hi Stefan,
Thanks very much for the details.  Very helpful.  Resolved a few
uncertainties for me which is great.  I did have the pins set up as you
suggest, and it helped me to discover that the adaptor board I was given
(free) from lcdlcd.com was mislabelled, and had pins 32-55 labelled as
25-50.  WIth all the pin mappings I made to get from ESP to display I
managed to convince myself it was all OK, and of course it wasn't.  Never
trust a freebie - it is worth less than you pay for it!

Nevertheless reconfiguring made no difference.  I suspect the two displays
I have tried out are fried.  Who knows what I have been feeding them.  I
have one spare display to burn now, and I think I will get the power
supplies in spec. before I blow another one.  I shall report back in due
course.

Really appreciate your help, and I'm sure the above documentation you
provided will help others.

More later,
mik[at]blackhole[dot]com

dreamy1, 2020-04-15:
Hi Mik,
does this lib support ESP32? Maybe only ESP8266 is supported, did you get any compiler issues?
Power supply of 3.3V for the 2.8V and 1.8V rail is ok, i have checked it double.

dreamy1, 2020-04-15:
Just an idea...maybe the SPI clock frequency of the ESP32 is too high.

mik00, 2020-04-16:
I have just torn it all apart and reconstructed the ratsnest from scratch.
I seem to have ended up with exactly the same configuration, so I'm more
confident about my pin assignments..
No compiler issues, and the right kind of stuff is coming on the scope... I
think.  The timings and signal polarities match up with the specification
document.

I have CS(white), SDA(grn), and CLK(orange) on the scope.
[image: IMG_20200415_171843.jpg]
The spec specifies a minimum clock cycle time of 66nS.  I'm showing 320nS
which is ~3MHz.  I don't think that can be too slow.

*Incidentally, does the LED turn on if I supply the correct voltages to
LEDK and LEDA (and nothing else is connected), or does the software have to
turn it on?*

I must have klutzed a connection somehow.  Everything looks right, and
matches your advice I believe.  I have beeped out the wiring as far as the
SMT socket.  I'm just reluctant to connect my last display until I have
exhausted all other options.

Thanks for your help.  I'll let you know when I finally figure it out.
This sucker is not going to beat me.

//Mik

dreamy1, 2020-04-16:
The backlight connections are on 1+2 (GND) and 23+24 (VCC), you have to place a series resistor for it to allow maximum specified 40mA@3.3V. I would recommend a 15 ohms resisstor. 
This voltage should be applied external and not from ESP32 because a portpin can't deliver this current. 
So the backlight is not software controlled out of the box, if you want to realize this, an additional hardware is required (e.g. a transistor).

mik00, 2020-04-16:
Bingo!

With your assistance everything was right except for the soldering.  In
exasperation I put it under the microscope.  I could just glimpse a tiny
solder ball from solder paste on the inside of the connector.  I popped it
off with a pin, and *voila*.  I can't believe that something so tiny could
make a difference, but it immediately solved the problem.   Soldering!!!
Duh.

https://photos.app.goo.gl/zo4u1787davwT2RYA

Obviously there are a few constants that are not quite right yet, but today
I am smiling!

Thanks big-time again Stefan.  I will pay it back, or forward one day.

(Just interested... where are you located in the world?  I'm in SF)

Mik

dreamy1, 2020-04-16:
Hi Mik,

congrats!!! Looks great and it seems there are only some minor things to do...just play with this entry in the ST7789.h:
 #define ST7789_240x240_YSTART 40

Good luck!
Stefan

P.S.: SF = San Francisco? I am located in germany near Heidelberg :-)

mik00, 2020-04-16:
Yes, SanFran.
Notice my email in earlier msg, if you are ever out this way.
Heidelberg  - went there once, and did the philosophers walk along the
hill.  Lovely town.  Beer destroyed me though!

Mik

dreamy1, 2020-04-16:
Wow, thats great...the philosophers walk i know very well :-)

But i was in SF too many years ago - for me, it is one of the most beautiful cities of the world and i was in many points of the world. Fisherman's wharf, Hawk hill....wonderful and i plan to come back somewhen.... :-)

Stefan

mik00, 2020-04-16:
Well - look me up if you come back.  I owe you a beer or three.

So will your code work with "h/w SPI" on the ESP32?  I
assume  Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST, TFT_CS); //for
display with CS pin
and use default SPI pins?

dreamy1, 2020-04-16:
Thanks for the beer in advance - i will come back to you :-)

I think hardware spi doesn't work because the special 9 bit solution is software bit banging. Myself i don't have this round lcd and i have used AVR only, not ESP - but with the two ST7789 lcd i use the code works fine.

mik00, 2020-04-16:
Indeed I can't make it work.
But I'll crank up the speed on the normal SPI and see how fast I can drive
it.

dreamy1, 2020-04-16:
Should be not a problem on an ESP to get more speed for the lcd :-)

deirvlon, 2020-04-18:
Hello @mik00,

> I recommend that if you have Arduino UNO first check displays with it, then apply to ESP32. At the same time, you can be sure that your wiring is correct. 

**This is the schematic of the screen that I am using:**

[image]

**The test on Arduino UNO - It is slow, but you can prove that you wired correctly ) :**

[image]

mik00, 2020-04-18:
Thanks for the good suggestion and help.  Solved.

mik00, 2020-04-18:
SUMMARY OF THINGS LEARNED ABOUT USING THIS DISPLAY WITH ESP32:
Stefan/dreamy1 is a helpful fellow!
I used a display described as follows:

> 1.3 inch 240x240 round IPS SPI tft lcd display with CTP for smart watch

SOURCE
I got it from Alibaba, but there are [ebay ](https://www.ebay.com/i/112800854794)sources with, and without touch screen.

DETAILS
This display does work with an ESP32.
The single ESP 3v3 pin **will** drive the 1v8, 2v8 **and** LED supplies enough to run the test program anyway as long as the ESP is actually producing 3v3!
The LED supply (LEDK/LEDA) seems to draw 20-30mA, but miraculously the ESP32 doesn't get upset driving the display backlight.
I have a 15 Ohm resistor between LEDK and 3v3.  (Thanks Stefan)
I used [Stefan's driver](https://github.com/ananevilya/Arduino-ST7789-Library/tree/spi9bit)
My ESP32 pin assignments were:
```
#define TFT_DC    8
#define TFT_RST   9 
#define TFT_CS    10 
#define TFT_MOSI  11  
#define TFT_SCLK  13
```  
The precise instantiation call I used was:
`Arduino_ST7789 tft = Arduino_ST7789(-1, TFT_RST, TFT_MOSI, TFT_SCLK, TFT_CS);`
I also had to change one line to get the display buffer in the write place on the screen (Thanks again Stefan)
`ST7789.h: <line 40> #define ST7789_240x240_YSTART 40`

FOR THE RECORD
Ultimately it appears my problems were totally unrelated.
A small ball of solder-paste in the connector.
A "free" adapter board from another display vendor with incorrectly labelled pinout (Duh!).
I suspect I connected every power pin to every display pin and the display still works fine! :-)

dreamy1, 2020-04-18:
Thanks for the feedback Mik :-)

Enjoy your lcd!
Stefan

JL325, 2020-06-06:
Did you guys know why my Arduino Uno R3 can't work on this code, but Mega2560 can make it?

mik00, 2020-06-06:
My first stop would be a 'scope.

M

JL325, 2020-06-08:
> Hello @mik00,
>
>
> **This is the schematic of the screen that I am using:**
>
> [image]
>
> **The test on Arduino UNO - It is slow, but you can prove that you wired correctly ) :**
>
> [image]

Hi Deirvlon,

Could you know why the UNO can't work for this code?

My display is AML130A2402A that using ST7789v.
I can't work on Uno and Pro mini is that seems doesn't initialize because the monitor just shows "Hello! ST7789 TFT Test".

but for the mega2560 is all work well.

deirvlon, 2020-06-18:
@JL325 Hi, it should work on UNO. I definitely recommend to check the wiring. 
Or try with other Arduino UNO.

deirvlon, 2020-06-18:
[image]

Also, be careful 
`//using hardware SPI (11, 13 on UNO; 51, 52 on MEGA; ICSP-4, ICSP-3 on DUE and etc)`

Try to connect using this configuration
@JL325

JL325, 2020-06-18:
@deirvlon Hi, I've tried two UNO device and I've checked OK with the wiring.
Do you mean that I try using the hardware SPI interface?

Thank you

deirvlon, 2020-06-18:
This is the function.
[image]

try after changing **YSTART 0** to **YSTART 40** in **_Arduino_ST7789.h_**

#define ST7789_240x240_YSTART 0
to
#define ST7789_240x240_YSTART 40

deirvlon, 2020-06-18:
[Test.zip](https://github.com/ananevilya/Arduino-ST7789-Library/files/4800498/Test.zip)

Display and  UNO not with me right now and I forgot most things about the display and its connections. This is the slightly changed version of this library.  I do not remember I have changed something in it to work or not. But this library is the latest version that I worked on. Try this one I hope it can solve your problem.

btukacs, 2020-08-06:
Any mod recommendations for ST7789V 240*135 display?

deirvlon, 2021-12-28:
I think it is the same with with 240x240 just change the screen size in the code @sigi77777