ananevilya/Arduino-ST7789-Library #5: does not work in hardware SPI mode

vsmelov, 2019-02-09:
Thanks you for a great library!

I connected my display to Arduino
```#define TFT_DC    4
#define TFT_RST   3 
#define TFT_MOSI  11   // for hardware SPI data pin (all of available pins)
#define TFT_SCLK  13   // for hardware SPI sclk pin (all of available pins)
```
so software SPI mode works
```
Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST, TFT_MOSI, TFT_SCLK); //for display without CS pin
```

but this does not 
```Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST); //for display without CS pin```

What can I check?

vsmelov, 2019-02-09:
Now it takes 2706ms to fill display black, what is quite slow, I think.

Can you please also say, where can I change frequency settings?

vsmelov, 2019-02-09:
changed a line in Arduino_ST7789.cpp from
```    mySPISettings = SPISettings(24000000, MSBFIRST, SPI_MODE2);```
to
```    mySPISettings = SPISettings(24000000/4, MSBFIRST, SPI_MODE2);```

Works like a charm now! Thank you!

Ratthanin, 2019-03-02:
Thank for great ST7789 Library, it works great with Nodemcu V1.0 connect to Hardware SPI
I just wonder why your library work but not Adafruit_7789 library.
so lucky I found your library here.

closed 2019-05-24 by ananevilya