ananevilya/Arduino-ST7789-Library #6: support for a display with 320x240 pixel resolution

DL8EBD, 2019-03-18:
Hi,
I use your library for 3 wire SPI (9bit) and it works great - but now I have a TFT with 320x240 pixel resolution.
Please, can someone help me to change the library so that I can use the higher resolution?
Thank you!

ananevilya, 2019-03-18:
I think at first you need to change this lines in **Arduino_ST7789.h** file to different resolution:
```
#define ST7789_TFTWIDTH_240 	240
#define ST7789_TFTHEIGHT_240 	240

#define ST7789_240x240_XSTART 0
#define ST7789_240x240_YSTART 0
```

DL8EBD, 2019-03-18:
Hello,
thanks for your fast answer, but it has no effect if I change WIDTH or HIGH to 320!
Any idea?

ananevilya, 2019-03-19:
Oh, maybe I know what's wrong. An error in initialization code. 
See lines 26 - 32 in **Arduino_ST7789.c** file. Resolution are hardcoded and set to 240 pixels on each dimension. I need to fix this. Could you change digits **240** to **ST7789_TFTWIDTH_240** and **ST7789_TFTHEIGHT_240** respectively and to check again?

DL8EBD, 2019-03-19:
ok, I think the solution was this:
_void Arduino_ST7789::setAddrWindow(uint**16**_t x0, uint**16**_t y0, uint**16**_t x1, uint**16**_t y1) {_
because 320 width is 16bit...

ananevilya, 2019-03-19:
Ok, thanks, I will fix **setAddrWindow** method too.
Have you tried to change hardcoded with and height? Or just changed the method parameters?

DL8EBD, 2019-03-19:
yes I changed the hardcoded with and high

ananevilya, 2019-03-19:
Thanks, I will fix this.

closed 2019-06-24 by ananevilya

KianYeo, 2019-09-10:
Hi. I am using a ST7789V 320 x 240 display. I have made the changes as in the comments above, but my display is not displaying correctly still. Its only filling up 240x240.

These are the changes I made:

In graphicstest.ino:
_tft.init(240,320);_

In Arduino_ST7789.h
_#define ST7789_TFTWIDTH 	240
#define ST7789_TFTHEIGHT 	320_

Am I missing something? Thanks in advance.