ananevilya/Arduino-ST7789-Library #7: Can it be used on NodeMCU?

MY201314MY, 2019-03-26:
I hava a ESP8266 board(Also called nodemcu).And i installed it on Arduino IDE,unfortunately it doesn't work. I think the library may be just bases on AVR microchip.

ananevilya, 2019-03-26:
I haven't tested with nodeMCU yet, but i think it should work. And one guy said that it works on esp8266.
https://github.com/ananevilya/Arduino-ST7789-Library/issues/2#issuecomment-453866275

MY201314MY, 2019-03-26:
Thakans for your Timely reply
There're many ways to program with NodeMCU,does the man use Arduinot?If so,the smart man may modify someting that I ignored.
The following is my warning ：
**Arduino-ST7789-Library-master\glcdfont.c:2:20: fatal error: avr/io.h: No such file or directory** 
 **#include **<avr/io.h>**
I have've nerver used the file with NodeMCU.Is the file necessary?(At least I think so it is.)

ananevilya, 2019-03-26:
This library for Arduino IDE only.

closed 2019-05-24 by ananevilya