ananevilya/Arduino-ST7789-Library #16: Can't work on Arduino Uno or Arduino pro mini, but Arduino mega2560 is OK.

JL325, 2020-06-07:
Hi Ananevilya,

I don't why I Can't work on Arduino Uno or Arduino pro mini for this code, but Arduino mega2560 is OK.

ananevilya, 2020-06-07:
> I don't why I Can't work on Arduino Uno or Arduino pro mini for this code, but Arduino mega2560 is OK.
Hi, what is the issue? 
Which type of display do you use? Which mode?
And what do you mean when you say "can't work"? Can't compile or can't init display or anything else?
Please explain in details.

JL325, 2020-06-08:
Hi Ananevilya,

I'm sorry, I didn't explain clearly.
My display is AML130A2402A that using ST7789v.
and what I said that can't work on Uno and Pro mini is that seems doesn't initialize because the monitor just shows "Hello! ST7789 TFT Test".

but for the mega2560 is all work well.

ananevilya, 2020-06-08:
I think you need to contact @deirvlon and check the wiring.
Please look at this comment https://github.com/ananevilya/Arduino-ST7789-Library/issues/3#issuecomment-615837702
The same display was tested with Arduino Uno and work as expected.